sinara-hw/Sampler #2: Sampler v2.0 noise measurement results

sotirova, 2018-03-29:
**Noise measurement**:

Configuration:
- Sample rate: 1/12us
- 10k samples per measurement, averaging over 100 measurements
- looking at channels 6 and 7. 6 has the 50R termination on, 7 has it off
- looking at noise as a function of frequency for different gains.

Results:

|gain | noise w/ termination | noise w/o termination|
|---|---|---|
|1 | 1.78LSB (544.3uV) | 1.75LSB (535.6uV) |
|10 | 1.84LSB (56.2uV) | 3.09LSB (94.4uV) |
|100 | 3.47LSB (10.6uV) | 26.02LSB (79.4uV) |
|1000 | 13.87LSB (4.2uV) | 206.3LSB (63.0uV) |

[image]

[image]

Expectations:
- ADC transition noise is 1.5LSB RMS, which should limit the low gain noise measurement
- BW of ADC driver is 1.1MHz.
- InAmp noise model:
  - for G=1: 45nv/rtHz * sqrt(1.1MHz) * 1.6=72uV For G=1, this gives 0.23LSB*gain, lower for higher gains
  - Noise current BW is presumably 1/(2 * pi * 200k * 82pF) so 10kHz
  - No termination: 5pA/rtHz * 100k * sqrt(10kHz) * 1.6 = 80uV = 0.26LSB*gain
(actually, that's not totally correct, because the RFI filter reduces this slightly...)
  - Termination: 5pA/rtHz * 2k *sqrt(1.1MHz) * 1.6 = 16uV, so negligible

- ADC driver:
  - Johnson noise from 20k resistors is 18nV/rtHz so negligible
  - OpAmp is low noise compared to InAmp

So, expected noise is:

|gain | noise w/ termination | noie w/o termination|
|---|---|---|
| 1 | 1.5LSB | 1.5LSB |
| 10 | ... | ... |
| 100 | ... | ... |
| 1000 | ... | ... |

Comments:

- Spurs are most likely due to pickup.
- Adding/removing the two screening cans doesn't make a noticeable difference to the data
- Running from a linear PSU/wall wart doesn't make a difference, nor does connecting the linear PSU -ve output to GND.

hartytp, 2018-03-29:
Expected noise (summing InAmp vn + in and ADC contributions):

|gain|noise w/ termination|noise w/o termination|
|---|---|---|
| 1 | 1.5LSB (460uV) | 1.5LSB (460uV) |
| 10 | 1.6LSB (50uV) | 2.6LSB (80uV) |
| 100 | 4.8LSB (14.7uV) | 21LSB (64uV) |
| 1000 | 14.4 LSB (4.4uV) | 206LSB (63uV) |

Done on paper, so may contain mistakes ;) but suggests that the numbers we see are consistent with datasheet values.

- For no-termination, taking the current noise BW as 1/(2*pi*200k*82pF) = 10k

PS NB factor of 1.6 in BW above should be *inside* the sqrt.

hartytp, 2018-03-29:
Note also that we decided to increase the ADC driver BW to 1.1MHz for the v2 design. That makes the noise worse for higher gains than in the v1.0 design. The motivation for this was some simulations I did for laser intensity servo loops, where I found that increasing the ADC BW gave a decent improvement in servo BW/settling time.

Personally, I think the extra BW is worth the increased noise (which is still very low for the applications I have in mind). But, if anyone feels that we should tweak the BW for the v2.1 design, I'm happy to reconsider. If we do decrease the BW, we could increase the capacitor before the ADC, which might help to reduce sampling transients (although, we don't currently have any evidence that these are a problem in the current design).

sotirova, 2018-04-16:
**Bandwidth measurement**

We measured the bandwidth as a function of gain for both small and large input signal - ou measurements are consistent with what we expect to see from the design so everything seems to be working fine. 

The large-signal bandwidths are very similar to the small-signal bandwidths. 

[image]

[image]

Method:

- 1e4 samples
- sample rate 1/1.26e-5 Hz
- sampling channel 7 (no termination) driven by sinusoid from Keysight 33500B generator
- Small-signal BW measured using 2Vpp/gain
- Large-signal bandwidth measured using 15Vpp/gain

hartytp, 2018-04-18:
List of tests:
- [x] Noise spectrum with 0V inputs (all gains, w + w/o testing)
- [ ] Verify that noise does not depend on input voltage, using batteries as a voltage source (G=1, G=1k)
- [x] Small + large signal BW (all gains)
- [x] Cross-talk (https://github.com/sinara-hw/sinara/issues/489)
- [x] Temp co? Ticked off as not worth measuring. IMHO, feel free to untick if you want this/can be bothered to do it!
- [x] CMMR Ticked off because we measured this on the V1 design and the AFE hasn't changed, so I'm happy trusting that data.
- [x] Harmonic content at some fixed frequency (all gains, large signal?). Ticked off because I don't think this is worth measuring.

hartytp, 2018-04-18:
(baiscally, I'm happy that this board is now well tested, and the other measurements can all be finished off later as and when people really need them!)

Thanks again to all involved in the testing.

closed 2018-04-27 by sotirova