sinara-hw/Sampler #3: gain error

jordens, 2018-05-17:
I am seeing a reading of 5V for 4.875V input and `0x7fff` for 9.76V input (pretty much the magic 1024/1000 error). I can't explain it from the schematics or the datasheets.
@hartytp @gkasprow have you also seen that?
Sampler/v2.1

gkasprow, 2018-05-17:
Was it the same in previous releases/batches?
I measured the DC response on rev1.0 and the difference was caused by the value of the RC filter which caused systematic error. This was fixed in rev 2.
I did not measure the DC response in rev2.x

gkasprow, 2018-05-17:
@jordens I assume you meant four references, REFOUT1..4. Or maybe REF_P2V5 and REF_P2V0?
Did you check the common mode of the ADC input? Maybe for some reason the DC offset appeared.
Please check also REF_P2V5 and REF_P2V0 on pin 1 and 7 of IC5. If you already checked, please check REFOUT1..4.

hartytp, 2018-05-17:
I don't think we saw that, no (although, I didn't do the testing myself so I can't be 100% sure).

jordens, 2018-05-18:
On one board REFOUT1,2 are 4.87V, REFOUT3,4 are ~4.9V. On another all four are 4.95V. All are definitely below 5.000V.

gkasprow, 2018-05-18:
So this explains the observation.
Can you measure voltage just at the C12 or pin 8 of ADC?

gkasprow, 2018-05-18:
Look at this voltage with scope as well. There is very little chance that reference oscillates.
I will recreate the issue but not earlier than Tuesday.
The ADC reference can be applied in two ways.
First - delivered to the REFOUT1..4, but then R10 must be removed (REFBFEN shorted to GND) and reference applied directly.
Second - delivered to the REF pin where it overrides internal reference. Internal reference has 250uA output current so can be easily overrided by external one.
It looks like the internal reference buffers are not real rail to rail amplifiers.
Did you check if the supply voltage is 5V? If it is slightly lower, that may be an issue and should be increased even to 5.25V by modification of 1 resistor. In such case this issue could be not detected on previous boards.
Another solution is to disable the buffers and apply the reference directly.

jordens, 2018-05-18:
No oscillations that I can see on REF or REFOUT1...4. I measured the 5V reference at C12.
P5V0A is also below 5V (~4.95V). Adding ~300k in parallel with R29 fixes it.

jordens, 2018-05-18:
I don't really like VDD=5.25V. And I don't know whether REFBUFEN=0, REFOUT1...4=5V, and VDD~4.99V would work. We could just simplify everything and lower the AFE gain and go with REF=4.096V. That also avoids complaints about "the ADC not doing 10.000V" and makes the math easier. Was/is the 1dB THD and 0.5dB SNR gain really worth it?

jordens, 2018-05-18:
I'll leave open until the fix has been committed to the repo or until this has landed on an errata list.

jordens, 2018-05-18:
I see you just did that. ;) Closing.

closed 2018-05-18 by jordens

gkasprow, 2018-05-18:
5.25 is still within operating range and far below maximum voltage (6V). Another option is to connect the REF to all ref outputs.
Since many Samplers were produced, changing the reference value now may cause frustration at some point. We would have to indicate it on the panel.

jordens, 2018-05-18:
5.25V is at the end of the specified range for VDD.
If you connect a 5V REF to all REFOUT, are you sure that will work with e.g. VDD=4.99V (what I wondered above)?
Supporting two different VREF in the code is not that hard. It's probably less inconvenient than running around and patching all Sampler boards with the issue.

gkasprow, 2018-05-18:
To change reference, you need to replace the chip anyway.
That's true, taking into account tolerances of resistors and regulator reference, one can exceed 5.25V. We would need to use 0.1% resistors. VADJ tolerance is 12mV (roughly 1%) and set 5.2V to make sure 5.25 will not be exceeded.
Direct connection needs testing, but it's easy to be done.
But probably 4.096V range would be more stable in long-term.
It would be hard to achieve +/-10V range with 4.096 reference due to missing matched resistive dividers that provide such division rate.

jordens, 2018-05-18:
Yes. More a lesson for the future. I am OK with 5.2V for now.
I meant +- 10.24 V range (2.5 gain, easy conversion math, and +10 V strictly included, also what many other products do).

hartytp, 2018-05-18:
@jordens Thanks for tracking this down!

I wonder why we didn't see it on the board we tested. Was there a change between v2.0 and v2.1 that affects this? Did we get lucky with component tolerances giving us a higher supply voltage on the board we tested? Or, did we just miss it? Anyway, I'll look at it when I'm back in the lab.

Assuming the dividers work out okay (someone needs to sketch out what they would be), I agree that going to a +-10.24 span (4.096Vref) sounds like the right idea. For the time being, I'll apply the same patch as @jordens to increase the supply voltage to 5.2V.

gkasprow, 2018-05-18:
We can move to 1/5 gain easily, we have:
2x5k and 2x10k in one package.
At the moment we have gain of 10+10 /5 = 4
We can combine 10+10+5/5 = 5.
We can make it in next revision.

hartytp, 2018-05-18:
@gkasprow what dividers would you use on the reference voltage to create the common-mode offsets?

gkasprow, 2018-05-18:
Exactly.

gkasprow, 2018-07-25:
fixed