sinara-hw/Sampler #9: v2.2 review

gkasprow, 2018-07-25:
@hartytp @jordens I want to make production run of v2.2
Please have a look at the [schematics](https://github.com/sinara-hw/Sampler/releases/tag/Sampler%2Fv2.2rc1)
I modified the ADC range, input stage gain and reference generation.

hartytp, 2018-07-25:
Thanks Greg. I'll look at it with fresh eyes in the morning.

hartytp, 2018-07-25:
Before I forget though, there are a few issues in https://github.com/sinara-hw/meta/issues that are relevant and would be nice to fix now (e.g. if it's easy, it would be nice to swap the EEM connector ordering).

hartytp, 2018-07-26:
Relevant "meta" issues are:
- https://github.com/sinara-hw/meta/issues/3
- https://github.com/sinara-hw/meta/issues/2
- https://github.com/sinara-hw/meta/issues/1

Only https://github.com/sinara-hw/meta/issues/1 affects the schematic/layout.

gkasprow, 2018-07-26:
OK, I will swap the EEM connectors

gkasprow, 2018-07-26:
done.

hartytp, 2018-07-27:
Do you need to make any PCB changes for the EMC gaskets (e.g. ensure that mount holes are connected to PCB ground)?

Otherwise, looks good to me. Thanks!

gkasprow, 2018-07-27:
OK, I changed the holes to plated and connected to GND.

jordens, 2018-07-27:
Woa. Why exactly was that done? Is it needed and is this the right moment and place to do it?
* AFAICT this is the only board that has this. Now this is the only point where the crate connects to GND.
* It's irrelevant for for EMC shielding AFAICT. The slot between the board and the panel is still there.

hartytp, 2018-07-27:
@jordens my understanding is that we agreed that all EEM front panels should connect to ground at the PCB, rather than relying on the long inductive loop of the IDC to do it.

gkasprow, 2018-07-27:
Guys, I can add 0R not mounted by default.

gkasprow, 2018-07-27:
I will make these 0R mounted by default.

jordens, 2018-07-27:
I am ok with reliably connecting the front panel to GND. I just have a problem with doing it in one place through single a connection that does not look like it's meant to ground the entire crate: Does that angle bracket reliably connect the front panel, does it cut through the passivation? I'd like to first have a consensus how we want to "properly" ground the case and after that start connecting to the case in various places. I.e. I'd rather leave this connection open now.
Where are we relaibly connecting to the crate currently? AFAICT:
* Kasli SFPs: I don't think they get through the passivation reliably. Or does it work somewhat because the inside of the cutout is freshly cut?
* Kasli Power: No
* Crate to EEM front panel: Not cutting through the passivation on the non-shielded crates, yes on the shielded crates
* Zotino HD68 shield, Grabber shield, HD68IDC shield: probably not even pressing against it
* DIO-RJ45: same as Kasli-SFP

hartytp, 2018-07-27:
> I'd like to first have a consensus how we want to "properly" ground the case and after that start connecting to the case in various places. I

Absolutely agreed.

I'd like to try to get this consensus before we make more EEMs to avoid spreading bad design decisions further.

> I just have a problem with doing it in one place through single a connection that does not look like it's meant to ground the entire crate

When you say "ground the entire crate" you mean "join the Kasli/EEM ground to the crate ground", right?

> Does that angle bracket reliably connect the front panel, does it cut through the passivatedA? 

@gkasprow 

> Where are we relaibly connecting to the crate currently? AFAICT:

I can't remember the details now (@cjbe) but, I've had a few cases where EEMs have grounded the chassis in unexpected ways, causing surprising issues. Currently, the crate is neither properly isolated nor properly connected to the EEMs, which seems like the worst possible situation.

AFAICT, the best solution is to agree that going forwards that all EEMs should ground to the crate via their front panels.

hartytp, 2018-07-27:
> Where are we relaibly connecting to the crate currently? AFAICT:

NB As I said, we're preparing to buy a fresh batch of EEMs, so the plan would be to ensure that this fix is applied to all EEMs produced from now on. That way, it's not just sampler that does the grounding.

We can leave 0R resistors so users can isolate the grounds if they need to.

jordens, 2018-07-27:
All correct. But there are dozens of crates out there with the current configuration. If any user wants to use a new module we need to come up with a way to handle this properly.

hartytp, 2018-07-27:
Okay, so we agree that the long-term solution is connecting the FP to the circuit ground (modulo the issue you raise about checking that the way Greg is doing this will actually work)?

> But there are dozens of crates out there with the current configuration. If any user wants to use a new module we need to come up with a way to handle this properly.

Well, the current situation is bad in a few ways. I'm not sure that adding the grounding on new EEMs makes it worse. And, in any case, there are 0R resistors which can be DNPd

Personally, I'd advice users to find a way of grounding their Kaslis to their crates.

jordens, 2018-07-27:
These options to rework the board are not satisfactory solutions IMHO. It's undocumented, hard to get through to the board house or the end user without causing delays/misunderstandings. The default needs to be consistent and correct. The burden of proof is with the proposed new solution, not the status quo. Does this attempt at grounding the panel improve the situation? Given the current state, I don't see how it has a positive effect on safety, on EMI, on consistency, and grounding quality. The only areas that come to mind are EMI and ESD, right? Since nobody wants to run current through this connection, I don't see how the power grounding problem is affected or resolved. If those new Sampler boards won't be manufactured with the EMI lip panels systematically or if the angle brackets are not meant to or can't reliably connect the panel then the default should be DNP and once we have the full fleet updated we can switch the default BOM.

gkasprow, 2018-07-27:
I released production documentation with grounding resistors DNP. One can request their assembly during ordering or mount them later.

hartytp, 2018-07-27:
> The default needs to be consistent and correct. 

Yes

> The burden of proof is with the proposed new solution, not the status quo

Sure: I think we all agree that the current status quo was not properly thought through and is not a good solution. Changing hastily to a new solution without due thought just multiplies the problems. So, let's see if we can settle on something better...

Currently, the EEM system, including power supply, tends to get grounded through unexpected/surprising paths. e.g. pressure put on an SFP cage that connects it to a front pannel, or an RJ45 DIO (grabber connectors?). That leads to unexpected and ill-defined grounding paths -- potentially involving sensitive lab equipment -- which is the absolute worst situation.

The proposal is to attempt to tie the circuit grounds to the chassis ground via their FPs. Lots of chassis provide grounding bolts or other grounding solutions, which we'd tie to a suitable ground point (e.g. all our racks are grounded, as are our optics tables).

I see one of the key benefits of this as being that it provides an explicit grounding path which users can understand and design around, and which won't lead to surprises. That would be a bit improvement.

Users who don't want their circuit grounds tied to mains ground can, for example, use an un-grounded chassis (there are plenty of options).

> Since nobody wants to run current through this connection, I don't see how the power grounding problem is affected or resolved. 

As I said, my plan is to have all EEMs + Kasli with their circuit grounds tied to the chassis ground through their FPs, and then to tie the chassis ground to mains ground. That does provide a solution to the power grounding problem. Can you think of a better way of achieving that?

>  If those new Sampler boards won't be manufactured with the EMI lip panels systematically or if the angle brackets are not meant to or can't reliably connect the panel then the default should be DNP and once we have the full fleet updated we can switch the default BOM.

That's a reasonable point. My understanding is that the new EEMs will all be manufactured with EMI lips etc.

@gkasprow are you happy that this will work with the new front panels you will supply (inc EMI gaskets)? Will they provide a good ground connection between the PCB and chassis? Can you generally comment on how you plan to do this?

Also, if you have any experience with "standard" ways of solving these grounding issues in lab instrumentation, feel free to suggest other ideas!

hartytp, 2018-07-27:
@gkasprow please can we have the 0R resistors populated in the next batch of EEMs you're producing for us? Thanks!

hartytp, 2018-07-28:
Let me caveat that: @gkasprow please can we have the 0R resistors populated if and only if
1. You're satisfied that the new front panels (you're planning to use Schroff ones with EMC gaskets, right?) will make a satisfactory connection between the PCB ground and the chassis ground
2. No one can think of any reason why connecting the PCB ground to chassis ground would not be the correct thing to do in most cases and no one can think of a better grounding arrangement

Subject to the above, I'd also suggest that we make having these 0R resistors populated the default option for all new boards.

sbourdeauducq, 2018-07-28:
> we make having these 0R resistors populated the default option for all new boards.

Let's hear what @cjbe has to say about the grounding problems and "surprising issues" first.

hartytp, 2018-07-28:
The issues I'm referring to, which were already posted, are things like the RJ45 boards connecting the circuit ground to chassis ground atm. That lead to an odd case where when the 12V connector was mis-wired (and missing a ground connection) Kasli only powered on when an RJ45 board was plugged in (the RJ45 shield connected Kasli to the rack ground, which was electrically earthed). I've seen similar issues with the SFP cages on Kasli which can make intermittent contact with the FP in at least one version of the FPs we have.

NB If we defloat the SMAs on Urukul (as we've agreed to do IIRC) then those will tie Urukul's circuit ground to the PCB ground in any case.

@sbourdeauducq what's your point here? Are you arguing that our current grounding arrangement isn't unsatisfactory?

hartytp, 2018-07-28:
AFAICT, the only good reason that's been raised not to connect the FP ground to circuit ground is if it won't actually work properly (e.g. due to anodisation of the FP).

sbourdeauducq, 2018-07-28:
No, I just want to make sure that @cjbe's problems get addressed in this iteration, so that we end up with 2 different grounding schemes to deal with and not 3.

hartytp, 2018-07-28:
Another "surprising" issue I've seen with the current grounding arrangement (reported elsewhere, but repeated for those who weren't following that discussion).

Before I de-floated the clock input connectors on Kasli, I was driving a floating Kasli board clock input from a synth. That lead to the synth ESD protection tripping, which is something I take very seriously, as I've seen this kill 20k synths before. De-floating the SMA fixed the problem. However, if one does this without grounding Kasli via the chassis (or some other suitable means) then the only electrical ground is via the synth/clock SMA, which is not a good place to be in at all. To fix this, I ended up both defloating the SMA and adding a grounding strap to my Kasli to bond it to my main earthing ground.

All the grounding issues that I'm aware of our group having seen have been reported and discussed. AFAICT, they will all be solved by connecting the EEMs to a grounded chassis, which is effectively what we're doing now.

hartytp, 2018-07-28:
Another way of putting all this:
- In some cases it is necessary to isolate grounds for noise purposes
- however, the correct approach to this is not to isolate all grounds by default (what we did in v1 boards effectively), as that creates the problems we've seen
- instead, a better solution is to tie all grounds together by default, but to give users the option of isolating ground loops in the rare cases that it's actually required. This is what I'd like to see done in v2.0. eg leaving the plastic insulating washers in to allow us to float clock SMAs if we want to, but by default using 0R resistors to ground all SMAs to the chassis/PCB ground

hartytp, 2018-07-29:
Oh and the other "unexpected" grounding issue that crops up in the current design is when the USB cable becomes the only mains grounding point so things can behave differently depending on whether USB connected or not. E.g. the synth was issues only happened when I removed the USB cable

gkasprow, 2018-07-30:
Usually we provide central grounding at the supply rail on the backplane. Since we don't have backplane, such grounding should be done on Kasli or if we use backplane SMPS module, it should be there, just at the DC module output.
In addition, panels are usually connected with GND either directly or using parallel RC network.
In case of RC networ it is to mitigate DC ground-loops and provide short-rircuit for EMI so the RF power path is shorted to chassis as close to the entry as possible.
So the main grounding point is central, close to the supply and all others are to mitigate EMI issues.

hartytp, 2018-07-30:
> So the main grounding point is central, close to the supply and all others are to mitigate EMI issues.

With the new Schroff FPs you're ordering, will the barrel connector on Kasli ground the PSU to the chassis via Kasli's FP?

gkasprow, 2018-07-30:
It should be grounded after the supply passes via CMCs. So grounding at the entry is not best option. It's better to ground it at the panel fixing to the PCB

hartytp, 2018-07-30:
@gkasprow okay.

I'm still not clear though: are you satisfied that the front panels for the next batch of EEMs will provide a good ground connection between the PCB ground and the chassis ground? The old panels did not seem to do that.

gkasprow, 2018-08-04:
The old panels did not do that. New panels are designed to do that so that should be enough.

closed 2018-08-04 by gkasprow