sinara-hw/Sampler #25: missing protection diodes

gkasprow, 2024-01-14:
The ADC might be in danger when overdriven from the front end, which can quickly go up to +/- 10V with currents up to 65mA. The only limit is a 51Ohm resistor. This is usually enough to trigger latch-up in most CMOS ICs.
I suggest adding protection diodes at the input of the ADC.

hartytp, 2024-01-15:
Can that happen in practice? I thought the divider between the InAmp and the OpAmp should prevent an input overdrive from leading to the ADC being overdriven?

gkasprow, 2024-01-15:
You are right! My colleague is re-using part of ADC with the driver (but without a divider), and I noticed that ADC might suffer damage. But that's not our case.
When you set the input range to 1V and apply 10V, the InAmp will produce, in the worst case, 11.5V.
We have an OutAmp with a gain of 1/5 and an offset of 1.7066. This means that the max voltage ADC would see is 4.3V.
No worry, then.

[image]

closed 2024-01-15 by gkasprow