sinara-hw/Sampler #26: Sampler DIOT with MCX

gkasprow, 2024-03-20:
We plan to make new revisions of all modules supporting the DIOT connector (we will leave the EEM option as well)
DIOT is 6 HP, so there won't be a place for BNC.
Instead, we could include 8 BNC adapters
[image]

jordens, 2024-03-20:
The shields touching (Sampler is differential) and the BNC being bulkhead mount troubles me.

gkasprow, 2024-03-20:
BNC Micro is another option, but it is not very popular though.
Yet another idea is to make a simple 1U panel that would hold all BNC-MCX pigtails. One could mount it below the DIOT crate.

gkasprow, 2024-07-10:
We can also make an 8-channel SMA.
But since we have a 6HP panel, we can also fit 8 BNCs
[image]

kaolpr, 2026-03-19:
But in my understanding it was with respect to adapters. Then indeed it may be touching. Don't seem to be the case for SMA.

kaolpr, 2026-09-23:
If I'm not mistaken, the reason mentioned by @jordens relates to those adapters, if multiple of them are connected to the module they casings can easily touch what effectively connects N lines. However, this is not the case with SMA.

If there is no other reason to stick to BNC, we suggest moving towards SMA distributed on 6 HP front panel (with mezzanine to provide a comfortable distance between connectors).

gravelpitman, 2026-10-05:
Together with @CannaCardo we prepared the following front panel arrangement with SMA connectors:
[image]

What changed:

- **Channel LEDs:** the termination LEDs are now SMD parts (`LED_AVAGO_HSMC-C120`), each placed next to its own channel's connector. LEDs for the channels on the main PCB sit on the main PCB, and LEDs for the channels on the mezzanine sit on the mezzanine.

[image]

- **Mezzanine mounting:** the mezzanine is mounted on Würth SMT threaded spacers (M3, `9774120360R`) to speed up assembly. The board-to-board distance is 12 mm.

[image]

Feedback is welcome.